jptrsn/ESP32-mqtt-room #5: Guru Meditation Error: Core 1 panic

jptrsn, 2018-12-14:
Bluetooth/WiFi coexistence issue affects MQTT connectivity. When MQTT async library disconnects during a BLE scan, device crashes with core 1 panic.

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40172a5f  PS      : 0x00060a30  A0      : 0x801b54d8  A1      : 0x3ffde240  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000045  A5      : 0x002c5912  
A6      : 0x00000000  A7      : 0x0000059c  A8      : 0x8017c93c  A9      : 0x3ffde210  
A10     : 0x3ffdd5c0  A11     : 0x3ffc1170  A12     : 0x3ffdd620  A13     : 0x00000000  
A14     : 0x3ffdd644  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000038  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40172a5f:0x3ffde240 0x401b54d5:0x3ffde260 0x4016af81:0x3ffde280

```

tomhash2, 2019-02-18:
Maby I am wrong but isn't it problem with MQTT library? Other projects face very similar problems, ex.: https://github.com/256dpi/esp-mqtt/issues/1

jptrsn, 2019-02-18:
@tomhash2 No, this repository uses the AsyncMQTT library for the ESP32. The error only occurs when the use of a BLE scan causes the WiFi process to lose connectivity.

jptrsn, 2019-03-19:
This bug is mostly quashed, as far as possible with my limited C++ skills.

closed 2019-03-19 by jptrsn

ashish-iottive, 2021-04-01:
What is the solution to resolve this situation, I am also encountering the same the same situation.

`18:16:31.579 -> A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
18:16:31.579 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffc6364  
18:16:31.579 -> A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000000  
18:16:31.614 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
18:16:31.614 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
18:16:31.614 -> 
18:16:31.614 -> Backtrace: 0x400014fd:0x3ffcede0 0x40110085:0x3ffcedf0 0x40118eba:0x3ffcf100 0x40118ef6:0x3ffcf190 0x400dea96:0x3ffcf1d0 0x400d4042:0x3ffcf270 0x400df41d:0x3ffcf2c0 0x4008e68d:0x3ffcf2e0
18:16:31.651 -> 
18:16:31.651 -> Rebooting...
18:16:31.651 -> ets Jun  8 2016 00:22:57
18:16:31.651 -> 
18:16:31.651 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:16:31.651 -> configsip: 0, SPIWP:0xee
18:16:31.651 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:16:31.651 -> mode:DIO, clock div:1
18:16:31.651 -> load:0x3fff0018,len:4
18:16:31.651 -> load:0x3fff001c,len:1216
18:16:31.651 -> ho 0 tail 12 room 4
18:16:31.651 -> load:0x40078000,len:9720
18:16:31.651 -> ho 0 tail 12 room 4
18:16:31.651 -> load:0x40080400,len:6352
18:16:31.683 -> entry 0x400806b8
18:16:32.124 -> Build: Apr  1 2021 18:14:51
`