jptrsn/ESP32-mqtt-room #8: Integrate with esphome lib

rdehuyss, 2019-03-04:
Have you considered contributing your work to esphome?

I'm currently working on creating a multi sensor which has:
- pir sensor
- bme680 for temperature
- light (lumen) sensor
- smoke sensor
- air quality

I also wanted to add ble scanning but esphome currently doesn't support that. The combination of your project and esphome would be great. Any idea why not to integrate?

jptrsn, 2019-03-05:
I've not looked into esphome at all, as I use tasmota on my DIY devices. I can take a look, but it's likely that this may not work in conjunction with it, as the ble scan necessitates disabling/suspending WiFi connectivity for a short while, which causes all kinds of issues with other code that's expecting the WiFi connection to be maintained constantly.

BLE and WiFi coexistence on the ESP32 can be troublesome, as both radios share the same antenna, and cannot be used simultaneously.

However, I am considering adding support for various sensors. I've got some light sensors, as well as some HTU21D temperature/humidity sensors. What chip/module/breakout are you planning on using for air quality, smoke, or motion? I'm thinking about the new radar devices for motion, but it may make sense to use PIR instead, as it would be limited to the physical boundaries of the room in which it's placed.

jptrsn, 2019-03-19:
@rdehuyss It looks like esphome now supports BLE scanning. See https://esphome.io/components/esp32_ble_tracker.html

closed 2019-03-19 by jptrsn

josemabcn, 2021-04-22:
Dear jptrsn,

It means that we can use the mqtt room with ESPHOME?, and how it's possible to integrate?

Thanks in advance,

Jose

jptrsn, 2021-04-23:
@josemabcn You would need to set up your own device tracking, not using mqtt room. It would be a different set up, as each ESPHome node needs to be configured with the device IDs to track. I am not using any beacon tracking in my own environment any more, so I am just responding to issues here and maintaining the repo.