jptrsn/ESP32-mqtt-room #9: ArduinoJson problem

mredone, 2019-03-16:
Does anyone successfully run this on ESP32
I tried 4h compile this on ArduinoIDE or PlatformIO and have all time same problem
ArduinoJson.h no file or directory.
I installed by hand or with package manager and always the same error.

Please any advice.
esp32 is perfect for room-assistant . RPi0W Is overkill.

jptrsn, 2019-03-17:
This project cannot be uploaded from the Arduino IDE without installing the libraries yourself. I strongly recommend using platformIO, which should automatically install all the required libraries for you.

Can you copy and paste the error message from the platformIO terminal? I'll see if I can reproduce the problem here. I uploaded to all my devices a couple weeks ago without issue, so I'm curious about what's happening for you.

mredone, 2019-03-19:
Sorry for non answer. Today i send here output from terminal

jptrsn, 2019-03-19:
I've found the issue, it was a breaking change from ArduinoJson 5 to 6. I expect to have a new release with an updated dependency in the near future.

closed 2019-03-19 by jptrsn