jptrsn/ESP32-mqtt-room #11: ArduinoJSON^6 has not been found in PlatformIO Registry; distance uninitialized

nealeydee, 2019-04-27:
Hello. When building this with Platformio, I get a message stating that a required library has not been found in PlatformIO Registry:

```
Warning! Library `{'requirements': None, 'name': 'ArduinoJSON^6'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'ArduinoJSON^6'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
```
which in turn results in this error:
ESP32-mqtt-room.ino:29:25: fatal error: ArduinoJSON.h: No such file or directory

In platformio.ini defined in libs_deps I can see l ArduinoJSON^6 - and I suspect  that the ^6 is surplus to requirements? Can anyone confirm?

Anyway, cannot build with that^6, but can build by removing that. However, when I do that, I get other build errors:

```
D:/BLE/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:110:11: warning: enumeration value 'SYSTEM_EVENT_ETH_DISCONNECTED' not handled in switch [-Wswitch]

D:/BLE/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:110:11: warning: enumeration value 'SYSTEM_EVENT_ETH_GOT_IP' not handled in switch [-Wswitch]
D:/BLE/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:110:11: warning: enumeration value 'SYSTEM_EVENT_MAX' not handled in switch [-Wswitch]
D:/BLE/ESP32-mqtt-room-master/ESP32-mqtt-room.ino: In lambda function:
D:/BLE/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:366:48: warning: unknown conversion type character 0xa in format [-Wformat=]
Serial.printf("Progress: %u% \n", percent);
^
D:/BLE/ESP32-mqtt-room-master/ESP32-mqtt-room.ino: In function 'void reportDevice(BLEAdvertisedDevice)':
D:/BLE/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:297:37: warning: 'distance' may be used uninitialized in this function [-Wmaybe-uninitialized]
} else if (mqttClient.connected() && distance >= maxDistance) {
^
D:/BLE/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:172:8: note: 'distance' was declared here
float distance;
^
```
I'm not too worried about most of these, but reportDevice and specifically distance not being declared will prevent this from working, right?

Having built this with platformio and uploaded it to my ESP32, I can see the ESP device connected in Home Assistant, but impossible to get my devices presence to show up.

Any ideas?

jptrsn, 2019-04-27:
You're correct that the `^6` should not be required, as long as you've got at least version 6 of ArduinoJSON included, as there are breaking changes between 5 and 6.

The warning about distance is that the variable may be uninitialized, but I've not found any problems related to that issue.

To see what devices are detected, you can use the serial monitor to watch the logs. Setting a scan duration that is very short can prevent devices from being detected. I'm currently using a scan duration of 10 seconds, with a 30 second delay between scans.

If you see devices being logged in the serial monitor, the next thing to check is your mqtt server. Subscribe to the room presence topic and see if the messages are getting through. If you see the device reports showing up, them you'll know the issue is with your home assistant config.

jptrsn, 2019-06-25:
Updated `platformio.ini` with correct syntax. This issue should now be resolved.

closed 2019-06-25 by jptrsn