jptrsn/ESP32-mqtt-room #12: Missing Settings_d.h

roblocksrocks, 2019-06-27:
I cannot compile in either Platformio or Arduino due to missing settings_d.h in both programs, I have tried searching for the relevant library and cannot find any results.  I followed the recommendation of creating settings_local.h so is settings_d.h a library or a file that I should be creating?

jptrsn, 2019-06-27:
Apologies, that slipped into a commit at some point. I'll correct it. I use an array of lettered sensor nodes, so I have an array of config files, one for each detector I'm using. It's meant to be `settings_local.h` or you can point it to any local settings file. If you use the format `settings_***.h` your configuration won't be wiped out by pulling the next release of code.

roblocksrocks, 2019-06-27:
Fixed, I just copied my settings_local to settings_d.  Thank you very much!  If you're set up for Buy Me A Coffee or similar I'll happily contribute.

closed 2019-06-27 by roblocksrocks

jptrsn, 2019-06-28:
I'm not set up to receive anything, but if you'd like, you can throw a buck or two to [Canada Blood Services](https://blood.ca/en). Just click on _ways to donate_ in the menu.

Hope you enjoy the project!

roblocksrocks, 2019-06-29:
Just donated what I could, I'm not Canadian but it looks like a great cause!

jptrsn, 2019-07-01:
Thanks so much for your generosity! I really appreciate the support.