jptrsn/ESP32-mqtt-room #13: Not reporting manufacturer data.

roblocksrocks, 2019-06-29:
I love the software so far, and it seems like it'll be perfect for my uses, the only issue I'm running into is that it's detecting devices without a UUID and just manufacturer data but not reporting them.  I can see in the log it finds them but it only sends the data for devices with a beacon UUID over MQTT therefore BeaconScope works but my MiBand and Fitbit for example do not, despite being picked up.

jptrsn, 2019-06-30:
That's weird, I'm able to track my mi band no problem. I use the mosquito command line to watch the `room_presence/#` topic, then bring my band close to the detector and watch the feed for the distance that's changing. You'll need to use the Bluetooth hardware Mac address, rather than a UUID for non-beacon devices.

I guess it would be a good candidate for a help file. I'm working on better documentation using GitHub pages, I'll add it to the list of tasks.

roblocksrocks, 2019-06-30:
I'll more than happily be your test dummy.  I've so far tried it using the
hardware ID and a couple different formats of the MAC.  If there's a
particular way to format the MAC in the room_presence I can try it but if
the logs are to be believed it's the program is detecting it but not
reporting so I'm not sure Home Assistant is even getting the chance to see
it.

jptrsn, 2019-07-01:
Can you capture some data and paste it back to this issue? Specifically, I'd be curious to see what's being published to the room presence topic, as well as the telemetry reported by the device. There are a few possibilities, and the info will help us narrow in on the root cause.

The only parameter that determines if the device that the ESP32 has seen will be reported or not is the distance threshold you've set in your config. Any device that's within the distance will be published to the mqtt topic. In order to see the device, you'll need to have your home assistant config written to match the exact ID as reported in the mqtt topic.

Can you use mosquitto_sub to capture the data coming into the mqtt topic? Ideally, place the band right next to the ESP32 so we can see device ID by looking at the distance reported and picking the closest one. The command will look something like this:
```
mosquitto_sub -h 192.168.1.13 -u {{mqtt user}} -P {{mqtt password}} -i logscapture -v -t "presence_nodes/#" | ts
```

Here's what my sensor config looks like for my Mi Band. The ID is just the MAC address of the band, without the colons.
```
- platform: mqtt_room
  device_id: C80F104DC7EB
  name: 'Mi Band bt_room'
```

roblocksrocks, 2019-07-02:
So it turns out I just had massive interference, I don't know why it seems to be only affecting this device.  The following readings are from anywhere between touching to about 6 inches direct line of sight.  The MAC/UUID in questions is dd866196dd0b.  When it was about 6 inches away it dropped off completely, everything else was from the band resting on the device.
`Jul 02 04:18:24 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-58,"distance":0.84}
Jul 02 04:18:24 room_presence/bathroom {"id":"2f234454cf6d4a0fadf2f4911ba9ffa6-1-1","uuid":"2f234454cf6d4a0fadf2f4911ba9ffa6","rssi":-47,"major":1,"minor":1,"txPower":-59,"distance":0.1}
Jul 02 04:18:24 room_presence/bathroom {"id":"dd866196dd0b","uuid":"dd866196dd0b","rssi":-64,"distance":1.8}
Jul 02 04:18:35 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-59,"distance":1.01}
Jul 02 04:18:35 room_presence/bathroom {"id":"2f234454cf6d4a0fadf2f4911ba9ffa6-1-1","uuid":"2f234454cf6d4a0fadf2f4911ba9ffa6","rssi":-50,"major":1,"minor":1,"txPower":-59,"distance":0.19}
Jul 02 04:18:35 room_presence/bathroom {"id":"2f234454cf6d4a0fadf2f4911ba9ffa6-1-1","uuid":"2f234454cf6d4a0fadf2f4911ba9ffa6","rssi":-46,"major":1,"minor":1,"txPower":-59,"distance":0.08}
Jul 02 04:18:35 room_presence/bathroom {"id":"dd866196dd0b","uuid":"dd866196dd0b","rssi":-62,"distance":1.43}
Jul 02 04:18:44 room_presence/bathroom {"id":"2f234454cf6d4a0fadf2f4911ba9ffa6-1-1","uuid":"2f234454cf6d4a0fadf2f4911ba9ffa6","rssi":-46,"major":1,"minor":1,"txPower":-59,"distance":0.08}
Jul 02 04:18:44 room_presence/bathroom {"id":"dd866196dd0b","uuid":"dd866196dd0b","rssi":-62,"distance":1.43}
Jul 02 04:18:55 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-62,"distance":1.43}
Jul 02 04:18:55 room_presence/bathroom {"id":"2f234454cf6d4a0fadf2f4911ba9ffa6-1-1","uuid":"2f234454cf6d4a0fadf2f4911ba9ffa6","rssi":-46,"major":1,"minor":1,"txPower":-59,"distance":0.08}
Jul 02 04:18:55 room_presence/bathroom {"id":"dd866196dd0b","uuid":"dd866196dd0b","rssi":-61,"distance":1.27}
Jul 02 04:19:05 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-61,"distance":1.27}
Jul 02 04:19:05 room_presence/bathroom {"id":"2f234454cf6d4a0fadf2f4911ba9ffa6-1-1","uuid":"2f234454cf6d4a0fadf2f4911ba9ffa6","rssi":-46,"major":1,"minor":1,"txPower":-59,"distance":0.08}
Jul 02 04:19:05 room_presence/bathroom {"id":"dd866196dd0b","uuid":"dd866196dd0b","rssi":-57,"distance":0.71}
Jul 02 04:19:15 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-55,"distance":0.5}
Jul 02 04:19:25 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-54,"distance":0.41}
Jul 02 04:19:35 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-56,"distance":0.59}
Jul 02 04:19:45 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-58,"distance":0.84}
Jul 02 04:19:45 room_presence/bathroom {"id":"dd866196dd0b","uuid":"dd866196dd0b","rssi":-64,"distance":1.8}
Jul 02 04:19:55 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-55,"distance":0.5}
Jul 02 04:20:05 room_presence/bathroom {"id":"26fee7752b44","uuid":"26fee7752b44","rssi":-54,"distance":0.41}
`

jptrsn, 2019-07-02:
It looks like your Mi Band's transmit strength is turned way down, leading the distance calculations to be off by a considerable amount. This would likely end up with the device usually being outside the max distance threshold, and thus not reported.

As far as I know, the only solution would be to crank up your max distance setting. However, this may result in devices "jumping" between rooms, as the mqtt room component of home assistant doesn't handle this well. It may be possible to modify the broadcast signal strength using an app such as NRF connect, but that's where my knowledge of bluetooth low energy and GATT characteristics runs out.

roblocksrocks, 2019-07-04:
Thank you for the assistance, I did try changing the distance and it did indeed jump.  Unfortunately I was not able to adjust using NRF Connect, perhaps because it's not really a beacon device.  Maybe this issue is specific to my device, maybe it's a characteristic of the Mi Band 3 but regardless I'll go ahead and close the issue

closed 2019-07-04 by roblocksrocks