jptrsn/ESP32-mqtt-room #15: disconnect

xadieu01, 2019-07-10:


jptrsn, 2019-07-10:
This can happen when there's a problem with your mqtt server, or if you have two devices attempting to use the same mqtt client ID. If you've only got one ESP32 running then it's definitely not the latter.

I was able to reproduce the problem, and I've added a bit of reconnect logic to address the issue. It's part of a larger piece of work to migrate to the newest ESP32 platform. If you pull the latest and reflash your device, the problem should go away.

xadieu01, 2019-07-10:
I have only one esp32. I was reinstall Mosquitto broker addon and i reflash the newest firmware for esp32 but nothing change! What should i do now?

xadieu01, 2019-07-10:
[https://i1084.photobucket.com/albums/j406/xadieu01vn/Untitled_5.jpg](url)
The esp32 restart constantly

jptrsn, 2019-07-10:
What version of the mosquitto add-on are you using? I've run into many problems with anything above 4.0, with username and password failing to connect. This problem affects devices in my network running Tasmota, other custom code, as well as this project. However, given you're seeing the telemetry data coming in, I have other suspicions.

The constant restarts are curious. There's a reconnect timer, which is meant to handle MQTT disconnects with a minimum of downtime. This came about because of the antenna sharing between bluetooth and WiFi causing sporadic connection failures, occasionally getting the device into an unresponsive state. The timer will reboot the device eventually if it is unable to connect.

Try deleting the `.pioenvs` and `.piolibdeps` folders from the project directory, then rebuild and upload.

xadieu01, 2019-07-10:
My mosquitto version is 5.0, i don’t know how to downgrade version. Before i reflash esp32, I delete all source folder and extract the new folder

xadieu01, 2019-07-10:
Log mosquitto:
1562781491: Client esp32_a has exceeded timeout, disconnecting.
1562781493: New connection from 192.168.1.11 on port 1883.
1562781493: New client connected from 192.168.1.11 as esp32_a (p2, c1, k60, u'xadieu01').

Esp32 connect and then disconnect

jptrsn, 2019-07-10:
Try changing line 18 of `platformio.ini` to this:
```
lib_deps = ArduinoJSON@^6, ESP32 BLE Arduino@^1.0.1, AsyncMqttClient@^0.8.2, AsyncTCP
```

xadieu01, 2019-07-10:
Thank you! It work like a charm

s JuConnecting to WiFi...
[WiFi-event] event: 0Wifi Ready
[WiFi-event] event: 2STA Start
[WiFi-event] event: 5WiFi lost connection
Connecting to WiFi...
[WiFi-event] event: 4[WiFi-event] event: 7IP address:   192.168.1.11
Hostname:       esp32_a
Connecting to MQTT
Scanning...     Connected to MQTT.
Success sending message to topic:       presence_nodes/home
Telemetry sent
Scan done! Devices found: 3
5b4801f900fb exceeded distance threshold 3.860000
fc039f633249 exceeded distance threshold 2.800000
devices_discovered: 3
devices_reported: 1
Telemetry sent
Scanning...     Scan done! Devices found: 3
5b4801f900fb exceeded distance threshold 3.120000
devices_discovered: 3
devices_reported: 2
Telemetry sent
Scanning...     Scan done! Devices found: 2
fc039f633249 exceeded distance threshold 2.800000
devices_discovered: 2
devices_reported: 1
Telemetry sent
Scanning...     Scan done! Devices found: 3
5b4801f900fb exceeded distance threshold 3.120000
devices_discovered: 3
devices_reported: 2
Telemetry sent
Scanning...     Scan done! Devices found: 2
fc039f633249 exceeded distance threshold 2.800000
devices_discovered: 2
devices_reported: 1

jptrsn, 2019-07-10:
Keep an eye on it, I've still been seeing occasional drop outs without reconnection. Others users of the Async Mqtt Client library have reported similar problems, which only began appearing for me with the latest update to the espressif32 platform. Hopefully it'll be resolved soon and I can implement a proper fix, along with fixing issue #14, where over the air updates don't work.

closed 2019-07-10 by jptrsn

xadieu01, 2019-07-10:
I have another question. How can I find UUID of miband, major and minor

xadieu01, 2019-07-10:
Ok I will keep my eyes on that!

jptrsn, 2019-07-10:
For tracking non-beacon BLE hardware, you can just use the MAC address, without colons. See [the updated documentation](https://jptrsn.github.io/ESP32-mqtt-room/generic_ble.html) for more info.