jptrsn/ESP32-mqtt-room #23: Support for multiple discovered beacons in presence channel

akkord64, 2019-09-10:
Would be nice to display multiple devices discovered in the topic.

If there are multiple devices reported, you should report all the devices found.

akkord64, 2019-09-11:
Example message:

{
"1": {"id":"2f234454cf6d4a0fadf2f4911ba9ffa6","uuid":"2f234454cf6d4a0fadf2f4911ba9ffa6","rssi":-88,"major":1,"minor":1,"txPower":-79,"distance":2.18}
"2": {"id":"2f234454cf6d4a0fadf2f4911ba9ffa7","uuid":"2f234454cf6d4a0fadf2f4911ba9ffa7","rssi":-88,"major":1,"minor":1,"txPower":-79,"distance":2.18}
}

jptrsn, 2019-09-11:
All devices discovered by the ESP32 that are within the max distance setting will be reported. Take a look at the `mqtt_room` [documentation for Home Assistant](https://www.home-assistant.io/components/mqtt_room/). All logic to assess the state of each beacon reported via MQTT is performed by Home Assistant, not by the modules themselves.

closed 2019-09-11 by jptrsn