jptrsn/ESP32-mqtt-room #25: MQTT states retained

emouawad, 2019-09-25:
**Describe the bug**
By default it appears that MQTT messages are being retained - which is reflected of course in HA.
When the ESP32 is running it reports CONNECTED for example - now because retain is enabled even-though the esp32 is powered off MQTT still has the old value as CONNECTED 

**To Reproduce**
Steps to reproduce the behavior:
1. Power On
2. Wait
3. Power Off

**Expected behavior**
No retain or Configurable retain on MQTT States - so after a given retain_timout state should be cleared- therefore consider esp32 disconnected

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

jptrsn, 2019-09-25:
Only the connection messages are retained - this is by design. When connected over MQTT, the module will set its LWT message. Your MQTT server should see the ESP32 has disconnected and publish the "DISCONNECTED" payload itself. If you're not seeing this happen, the problem likely lies in your MQTT server rather than in the module itself.

The purpose of retaining the message is to ensure that Home Assistant is aware of the connectivity state after a reset. There are no periodic publish events of connectivity - only on connect or disconnect. Thus, retention of the message payload is required in order for Home Assistant to understand the correct state at time of boot.

I am not able to reproduce this issue in my Home Assistant instance.

closed 2019-09-25 by jptrsn

emouawad, 2019-09-25:
Maybe what happened is that the mqtt server went down before those ESPs - 
Later mqtt restarted and ESPs were still down

Can you confirm that in that case retained states no longer reflect the state of the ESPs

jptrsn, 2019-09-25:
You're correct, if the MQTT server goes down, the LWT message information will likely be lost. At that point, the retained states will not reflect the connectivity state of the ESPs. You'd need to reset them by publishing a null or empty payload with the retain flag.