jptrsn/ESP32-mqtt-room #26: Unable to Build/Compile.

maxi1134, 2019-10-26:
**Describe the bug**
When trying to build the Project, I get an error, tried with both VS and Atom. Both using the PlatformIO plugin

**Configuration**
#define ssid "Chilling Spot"
#define password ""
#define hostname "esp32_room_presence"
#define mqttHost IPAddress(192, 168, 34, 15)
#define mqttPort 1883
#define mqttUser "homeassistant"
#define mqttPassword "123456789"
#define room "living-room"
#define channel "room_presence"
#define scanInterval 15
#define singleScanTime 10
#define activeScan true
#define bleScanInterval 15
#define bleScanWindow 20

**Logs**
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.11.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
Converting ESP32-mqtt-room.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for ArduinoJSON library in registry
Found: https://platformio.org/lib/show/64/ArduinoJson
LibraryManager: Installing id=64 @ ^6
ArduinoJson @ 6.12.0 is already installed
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Arduino> 1.0.1
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.1
|-- <AsyncTCP> 1.1.1
|-- <ArduinoJson> 6.12.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Compiling .pio\build\esp32\src\ESP32-mqtt-room.ino.cpp.o
Archiving .pio\build\esp32\lib395\libAsyncTCP_ID1826.a
Archiving .pio\build\esp32\lib7b6\libAsyncMqttClient_ID346.a
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino: In function 'bool sendTelemetry(int, int)':
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:107:21: error: 'maxDistance' was not declared in this scope
  tele["max_dist"] = maxDistance;
                     ^
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:122:25: error: 'telemetryTopic' was not declared in this scope
  if (mqttClient.publish(telemetryTopic, 0, 1, teleMessageBuffer) == true) {
                         ^
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino: In function 'void WiFiEvent(system_event_id_t)':
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:209:19: error: 'LED_BUILTIN' was not declared in this scope
      digitalWrite(LED_BUILTIN, !LED_ON);
                   ^
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:209:33: error: 'LED_ON' was not declared in this scope
      digitalWrite(LED_BUILTIN, !LED_ON);
                                 ^
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:207:9: warning: enumeration value 'SYSTEM_EVENT_SCAN_DONE' not handled in switch [-Wswitch]
   switch(event) {
         ^
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:207:9: warning: enumeration value 'SYSTEM_EVENT_STA_CONNECTED' not handled in switch [-Wswitch]
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:207:9: warning: enumeration value 'SYSTEM_EVENT_STA_AUTHMODE_CHANGE' not handled in switch [-Wswitch]
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:207:9: warning: enumeration value 'SYSTEM_EVENT_STA_LOST_IP' not handled in switch [-Wswitch]
[... 57 lines cut ...]
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino: In function 'bool sendTelemetry(int, int)':
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:130:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino: In function 'bool handleMqttDisconnect()':
C:/Users/Maxi/Documents/PlatformIO/ESP32-mqtt-room-master/ESP32-mqtt-room.ino:173:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
*** [.pio\build\esp32\src\ESP32-mqtt-room.ino.cpp.o] Error 1

**To Reproduce**
Steps to reproduce the behavior:
1. Open Atom
2. Open the project through PLatformIo
3. Press on build.
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

closed 2019-10-28 by maxi1134

jptrsn, 2019-10-28:
@maxi1134 I assume you found the solution. Can you post the fix here so other users who encounter the problem can sort it out? 

Thanks!

maxi1134, 2019-10-28:
Of course!

I was using an old config files without some of the Declarations. (Led_Builtin and such).

Be sure to use new files for everything people! Otherwise you end up looking for a solution to an non-existing issue.

betuelrov, 2020-12-14:
Hi im running the same issue. Wich config files are you refering please

betuelrov, 2020-12-14:
haracteristics(std::map<short unsigned int, BLERemoteCharacteristic*>*)':
C:\Users\Betuel\.platformio\packages\framework-arduinoespressif32\libraries\BLE\src\BLERemoteService.cpp:246:89: warning: parameter 'pCharacteristicMap'
set but not used [-Wunused-but-set-parameter]
 void BLERemoteService::getCharacteristics(std::map<uint16_t, BLERemoteCharacteristic*>* pCharacteristicMap) {
                                                                                         ^
========================== [FAILED] Took 7.96 seconds ==========================

betuelrov, 2020-12-14:
Processing esp32 (platform: espressif32; framework: arduino; board: esp32dev)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jt
ag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting ESP32-mqtt-room.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Arduino> 1.0.1
|-- <ArduinoJson> 6.17.2
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.1
|-- <AsyncTCP> 1.1.1
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <Wire> 1.0.1
Building in release mode
Compiling .pio\build\esp32\src\ESP32-mqtt-room.ino.cpp.o
Archiving .pio\build\esp32\lib38e\libAsyncMqttClient.a
Archiving .pio\build\esp32\lib9d7\libUpdate.a
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino: In function 'void WiFiEvent(system_event_id_t)':
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_SCAN_DONE' not handled in switch [
-Wswitch]
   switch(event) {
         ^
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_STA_CONNECTED' not handled in swit
ch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_STA_AUTHMODE_CHANGE' not handled i
n switch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_STA_LOST_IP' not handled in switch
 [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_SUCCESS' not handled in
 switch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_FAILED' not handled in
switch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_TIMEOUT' not handled in
 switch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_PIN' not handled in swi
tch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_STA_WPS_ER_PBC_OVERLAP' not handle
d in switch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_AP_START' not handled in switch [-
Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_AP_STOP' not handled in switch [-W
switch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_AP_STACONNECTED' not handled in sw
itch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_AP_STADISCONNECTED' not handled in
 switch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_AP_STAIPASSIGNED' not handled in s
witch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_AP_PROBEREQRECVED' not handled in
switch [-Wswitch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_GOT_IP6' not handled in switch [-W
switch]
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:248:9: warning: enumeration value 'SYSTEM_EVENT_ETH_START' not handled in switch [
[... 15 lines cut ...]
Compiling .pio\build\esp32\lib1f3\WiFi\WiFiSTA.cpp.o
In file included from C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:36:0:
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino: In function 'void setup()':
src\Settings.h:14:18: error: '$MQTT_PORT$' was not declared in this scope
 #define mqttPort $MQTT_PORT$
                  ^
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:568:34: note: in expansion of macro 'mqttPort'
   mqttClient.setServer(mqttHost, mqttPort);
                                  ^
Compiling .pio\build\esp32\lib1f3\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32\lib1f3\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32\lib1f3\WiFi\WiFiUdp.cpp.o
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino: In function 'bool handleMqttDisconnect()':
C:/Users/Betuel/Desktop/ESP32-mqtt-room-master/src/ESP32-mqtt-room.ino:214:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Compiling .pio\build\esp32\libbdd\ESPmDNS\ESPmDNS.cpp.o
Compiling .pio\build\esp32\lib79b\ArduinoOTA\ArduinoOTA.cpp.o
Compiling .pio\build\esp32\libcfb\Wire\Wire.cpp.o
Archiving .pio\build\esp32\libFrameworkArduinoVariant.a
Compiling .pio\build\esp32\FrameworkArduino\Esp.cpp.o
*** [.pio\build\esp32\src\ESP32-mqtt-room.ino.cpp.o] Error 1
========================== [FAILED] Took 4.86 seconds ==========================

Please what im doing wrong...help!

jptrsn, 2020-12-14:
Please make sure you have followed the directions in the "Define your settings" section. https://jptrsn.github.io/ESP32-mqtt-room/#define-your-settings

betuelrov, 2020-12-14:
Thanks for the reply. Yes, it was a  settings issue. anyway now i'm stuck
at  mqtt_room setup at HA  whatever i try i can get a valid configuration
at configuration.yaml
Using this example with my data does not work

  - platform: mqtt_room
    device_id: "mydevicetotrackMAC"
    name: 'my device Room Presence'
    state_topic: 'room_presence'
    timeout: 60

away_timeout: 120

What can i do? Im very noob to this sorry for asking. Thanks in advance!

betuelrov, 2020-12-14:
By the way. binary sensor is working perfectly

jptrsn, 2020-12-17:
@betuelrov can you open a new issue for the troubleshooting of setting up mqtt_room? Take a look at the sections on the website for setting up home assistant (https://jptrsn.github.io/ESP32-mqtt-room/home_assistant) and for troubleshooting (https://jptrsn.github.io/ESP32-mqtt-room/troubleshooting).