jptrsn/ESP32-mqtt-room #28: Compilation code with VScode

yorchtf, 2020-02-14:
**Describe the bug**
I don't understand the problem, maybe the error is relation with ArduinoJson library/version.
`.pio/libdeps/esp32/ESPAsyncTCP_ID305/src/AsyncPrinter.cpp:48:11: error: 'panic' was not declared in this scope`

**Configuration**
```
//Replace with your Wifi SSID; example: #define ssid "MyWifi"
#define ssid "Ubuntu"

//Replace with your Wifi password; example: #define password "12345678"
#define password "V1nue$@R@mo$%5"

//Replace with a human-friendly host name. Must not contain spaces or special characters and be unique on your network
#define hostname "esp32_dormitorio_presence"

//Replace with your MQTT Broker address; example: #define mqttHost IPAddress(192, 168, 1, 195)
#define mqttHost IPAddress(192, 168, 1, 19)

//Replace with your MQTT Broker port; example: #define mqttPort 1883
#define mqttPort 1883

//Replace with your MQTT Broker user; example: #define mqttUser "homeassistant"
#define mqttUser "mqtt"

//Replace with your MQTT Broker password; example: #define mqttPassword "12345678"
#define mqttPassword "tfV1nue$@62010mq"

//Replace with the room name where the node will be placed; example: #define room "living-room"
#define room "dormitorio"

//Specify the LED pin. For most dev boards, this is GPIO2
#define LED_BUILTIN 2

// Logic level for turning the led on. Most boards use active low, meaning LED_ON should be set to 0
#define LED_ON 0

//Define the base topic for room detection. Usually "room_presence"
#define channel "room_presence"

//Define the topic for publishing availability
#define availabilityTopic "presence_nodes/" room

//Define the topic for publishing JSON attributes
#define telemetryTopic "presence_nodes/" hostname "/tele"

// Define bluetooth scan parameters
#define scanInterval 10 // Define the interval in seconds between scans
#define singleScanTime 15 // Define the duration of a single scan in seconds
#define activeScan true // Active scan uses more power, but get results faster
#define bleScanInterval 0x80 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing
#define bleScanWindow 0x10 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing

// Maximum distance (in meters) to report. Devices that are calculated to be further than this distance in meters will not be reported
#define maxDistance 10
```

**Logs**
`> Executing task: platformio run --target upload <

Processing esp32 (platform: espressif32; framework: arduino; board: esp32dev)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.11.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
[... 172 lines cut ...]
*** [.pio/build/esp32/lib83d/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o] Error 1
/Users/yorchtf/playground/domotica/ESP32-mqtt-room-ORIGINAL/ESP32-mqtt-room.ino: In function 'bool reportDevice(BLEAdvertisedDevice)':
/Users/yorchtf/playground/domotica/ESP32-mqtt-room-ORIGINAL/ESP32-mqtt-room.ino:409:18: warning: 'distance' may be used uninitialized in this function [-Wmaybe-uninitialized]
     Serial.printf("%s exceeded distance threshold %.2f\n\r", mac_address.c_str(), distance);
                  ^
/Users/yorchtf/playground/domotica/ESP32-mqtt-room-ORIGINAL/ESP32-mqtt-room.ino:283:8: note: 'distance' was declared here
  float distance;
        ^
================================================================================================ [FAILED] Took 6.86 seconds ================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.`

**Hardware Details**
If the bug is related to tracking a specific piece of hardware (e.g. iBeacon, phone, etc), please enter any relevant details about the hardware (such as firmware version, MAC address, etc).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Two months ago, the compilation was finished correct without problem with the same code.

devcpu, 2020-02-14:
Same errors here. Checked with older versions of my code from git. It seems to be an error in the espressif libs.

XYZVector, 2020-02-15:
Yep getting this also trying to compile against older Espressif versions and ESP Arduino platforms.

XYZVector, 2020-02-15:
Trying to remove all cached dependencies, and going back to older library versions, I think the latest Espressif 32 core nuked the panic function, and it also looks like the new nothrow exceptions were deleted. There is also a change to the addr_t now it has an enumeration for IPV6. I am not in a down situation, however I will spend about an hour trying different config versions. Then I got to get back to Valentines Day.

devcpu, 2020-02-15:
May be a solution
No clue why, but platformio adds ESPAsyncTCP to the libraries, but this lib is for ESP8266 only! For ESP32 you have to use AsyncTCP! It seems be be an error in LDF. After removing (uninstalling) ESPAsyncTCP from platformio it works again.
Better solution (tested) is to add following to platform.ini
lib_ignore = ESPAsyncTCP

yorchtf, 2020-02-15:
Ohhhh your are idea seems true... now I'm not in my house but It seems compile the code in the computer (I haven't got  a ESP32 in this moment):
```
.....
Compiling .pio/build/esp32/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32/libFrameworkArduino.a
Indexing .pio/build/esp32/libFrameworkArduino.a
Linking .pio/build/esp32/firmware.elf
Building .pio/build/esp32/firmware.bin
Retrieving maximum program size .pio/build/esp32/firmware.elf
Checking size .pio/build/esp32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.1% (used 62692 bytes from 327680 bytes)
Flash: [=======   ]  72.9% (used 1433970 bytes from 1966080 bytes)
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.```

I think that the last ERROR is becuase I haven't got a ESP32 now connect to my computer.... This night, I will test wit a ESP32

yorchtf, 2020-02-16:
@devcpu Thanks a lot, It worked!!!

closed 2020-02-16 by yorchtf

jptrsn, 2020-02-17:
Thanks @devcpu for the fix. I've applied it to the master branch of the project, users should not run into this problem in the future.