jptrsn/ESP32-mqtt-room #31: Static IP setup

giorgioarchi, 2020-02-22:
**Is your feature request related to a problem? Please describe.**
Thank you for this amazing software, I would like to configure a static IP

**Describe the solution you'd like**
I've tried to implement the standard "Static IP" code for ESP32 but i get:
[E][AsyncTCP.cpp:934]

jptrsn, 2020-02-24:
@giorgioarchi I'm not sure why this was closed without response - I've reopened it.

The current codebase does not support setting a static IP address. A workaround could be to use an address reservation on your router's DHCP server, ensuring that the ESP32 will receive the same IP address each time it connects. Would this be sufficient to meet your needs?

giorgioarchi, 2020-02-24:
Thank you so much for your reply, unfortunately i would like to set a static IP different from the DHCP pool.

It's just an idea for a future improvement i can deal with it, your code saved me a lot of headaches thank you!

closed 2021-01-25 by jptrsn