jptrsn/ESP32-mqtt-room #33: BLE Beacon Not Showing

mdframe, 2020-03-15:
**Describe the bug**
I have a BLE bluetooth beacon, https://www.amazon.com/gp/product/B07FC5FMHW/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1, which I have used with other ESP32 room presense software and it always shows as active.  This application is not showing this device and never detects it from what I can tell.  All of my existing Tile devices show however none of the devices I have listed above do.  Again, these were showing with other ESP32 room presence software on this exact hardware. 

**Configuration**
//Replace with your Wifi SSID; example: #define ssid "MyWifi"
#define ssid "PQxxxxxxxxxxxxxxr"

//Replace with your Wifi password; example: #define password "12345678"
#define password "Jxxxxxxxxxxxxxxx"

//Replace with a human-friendly host name. Must not contain spaces or special characters and be unique on your network
#define hostname "esp32_garage"

//Replace with your MQTT Broker address; example: #define mqttHost IPAddress(192, 168, 1, 195)
#define mqttHost IPAddress(192, 168, 1, 50)

//Replace with your MQTT Broker port; example: #define mqttPort 1883
#define mqttPort 1883

//Replace with your MQTT Broker user; example: #define mqttUser "homeassistant"
#define mqttUser "hassioxxxxxxxxxx"

//Replace with your MQTT Broker password; example: #define mqttPassword "12345678"
#define mqttPassword "Fxxxxxxxxxxxxxx"

//Replace with the room name where the node will be placed; example: #define room "living-room"
#define room "garage"

//Specify the LED pin. For most dev boards, this is GPIO2
#define LED_BUILTIN 2

// Logic level for turning the led on. Most boards use active low, meaning LED_ON should be set to 0
#define LED_ON 0

//Define the base topic for room detection. Usually "room_presence"
#define channel "room_presence"

//Define the topic for publishing availability
#define availabilityTopic "presence_nodes/" room

//Define the topic for publishing JSON attributes
#define telemetryTopic "presence_nodes/" hostname "/tele"

// Define bluetooth scan parameters
#define scanInterval 5 // Define the interval in seconds between scans
#define singleScanTime 5 // Define the duration of a single scan in seconds
#define activeScan true // Active scan uses more power, but get results faster
#define bleScanInterval 0x80 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing
#define bleScanWindow 0x10 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing

// Maximum distance (in meters) to report. Devices that are calculated to be further than this distance in meters will not be reported
#define maxDistance 5

sensors.yaml
- platform: mqtt_room
    device_id: '1804ed51934c' # Note that major version must match, but any minor version will be ignored
    name: 'Puck 1'
    state_topic: 'room_presence'
    timeout: 60
    away_timeout: 120

binary_sensors.yaml
  - platform: mqtt
    name: Esp32 Garage
    state_topic: "presence_nodes/garage"
    json_attributes_topic: "presence_nodes/esp32_garage/tele"
    payload_on: "CONNECTED"
    payload_off: "DISCONNECTED"
    device_class: connectivity

**Logs**
Nothing to show as this particular device is not being discovered by the ESP32.

**Hardware Details**
https://www.amazon.com/gp/product/B07FC5FMHW/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

**To Reproduce**
Place device near ESP32 to be read and the ID never shows in the MQTT queueu. 

**Expected behavior**
I should see the device id being read and displayed in the MQTT topic.  It't almost like the device is being filtered out but I am unable to determine why. 

I have been through the source code and have not seen any reason for this not to show on this particular device ESP32.  I have utilized this ESP for multiple tests with other firmware however this is the only one that I found that these devices are not showing.  So far, this is the big winner for me in for room presence and if I can get past this hurdle I will have several of these running in my home.  Both test devices right now are having the same issue whereby the device is not showing.  Both were utilized for other firmware tests and it previously reported without issue.

mdframe, 2020-04-05:
I identified the issue as being within the source code.  The BLE devices I have were being categorized as Eddystone, which is correct, however the structure of the code never set the distance variable which therefore never matched the criteria for being sent to the MQTT host. 

The following item shows the before and after of what I updated:

```
if (advertisedDevice.getServiceDataUUID().equals(BLEUUID(beaconUUID))==true) {  // found Eddystone UUID
		  //Serial.printf("is Eddystone: %d %s length %d\n", advertisedDevice.getServiceDataUUID().bitSize(), advertisedDevice.getServiceDataUUID().toString().c_str(),strServiceData.length());

		if (cServiceData[0]==0x10) {
```
Updated code:
```
if (advertisedDevice.getServiceDataUUID().equals(BLEUUID(beaconUUID))==true) {  // found Eddystone UUID
		  //Serial.printf("is Eddystone: %d %s length %d\n", advertisedDevice.getServiceDataUUID().bitSize(), advertisedDevice.getServiceDataUUID().toString().c_str(),strServiceData.length());

                // Updated distance variable for Eddystone BLE devices
		   BLEBeacon oBeacon = BLEBeacon();
		   distance = calculateDistance(rssi, oBeacon.getSignalPower());
		   doc["distance"] = distance;

		if (cServiceData[0]==0x10) {
```

If the original author means for this to operate in a different manner please let me know however after this update all of my Eddystone devices are working as expected.  

HTH someone else!

Matt

jptrsn, 2020-04-08:
Thanks for the tip. I've added the fix to the development branch, so it will be included in the next release.

commit 1b9070e1 (2020-04-08): Added distance calculation to Eddystone beacon to fix #33
closed 2020-04-08 by jptrsn