jptrsn/ESP32-mqtt-room #34: Question/Feature Request: Support of PIR and maybe other sensors

austwhite, 2020-05-10:
**Is your feature request related to a problem? Please describe.**
As the ESP32 supports many IO devices, I would like to utilize it for other functions, such as Motion and temperature monitoring using something like a DHT22 or Dallas DS18B20.
Ideally this would be selectable via a web interface on the device so it would feel less project like.

**Describe the solution you'd like**
To be able to utilize other sensors on the ESP32 while running ESP32-mqtt-room

**Describe alternatives you've considered**
If not integrated or able to be integrated in this project, it would involve having separate devices to perform this function

jptrsn, 2020-05-10:
Thanks for the feature request! I've begun working on integrating the HTU21D temperature/humidity sensor. However, I will probably not be building out a suite of sensor integrations - when I have the time, I will attempt to write a custom sensor for the esphome project. If I am able to get that operational, it will address all your needs for integrating other sensors, as well as potentially reach a wider audience.

I know it's frustrating to see all the functionality available on the ESP32, and to have it only using the bluetooth/wifi. However, the goal of this project is to provide basic functionality without the need for any external hardware or a soldering iron.

Cheers!

closed 2020-05-10 by jptrsn

austwhite, 2020-05-10:
If you can get a code to work in ESPHome for similar functionality, then that would be ideal as I use a few ESP32s using QuinLED boards for lighting.  That could be an ideal solution for me.

I appreciate your quick response :)