jptrsn/ESP32-mqtt-room #35: Not connecting to WiFi

uli6, 2020-08-12:
Hi guys, I'm trying to flash an ESP32 Wemos Lolin 32 OLED.

[This one](https://pt.banggood.com/ESP32-OLED-Module-ESP32-OLED-WiFi-bluetooth-Dual-ESP-32-ESP-32S-ESP8266-p-1148119.html?gmcCountry=PT&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=lijing&utm_campaign=ssc-ptg-0724-usw-all&gclid=CjwKCAjwps75BRAcEiwAEiACMW4x9A5KV0nJjVE_cCyIqd69HgIXR2qITX6sIdCu-l95ZjIZXZjWDhoCwEcQAvD_BwE&cur_warehouse=CN)

However, I'm receiving this information on serial monitor:

```
k⸮E⸮⸮⸮0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi...
[WiFi-event] event: 0

Wifi Ready
restarted
[WiFi-event] event: 2

STA Start
WiFi Time remaining: 1999
Connecting to WiFi...
[WiFi-event] event: 5

WiFi lost connection, resetting timer	
restarted
Connecting to WiFi...
[WiFi-event] event: 5

WiFi lost connection, resetting timer	
restarted
Connecting to WiFi...
[WiFi-event] event: 5

WiFi lost connection, resetting timer	
restarted
Connecting to WiFi...
[WiFi-event] event: 5

WiFi lost connection, resetting timer	
restarted
Connecting to WiFi...
[WiFi-event] event: 5

WiFi lost connection, resetting timer	
restarted
```

I'm using Visual Studio for MAC OS with Platform.io

----------------------------- Remove above -----------------------------
Hardware:

Board: Lolin32
IDE name: Platform.io
Upload Speed: 115200
Computer OS: Mac OSX

jptrsn, 2020-08-14:
That's odd, I haven't come across that before. Are you certain your WiFi credentials are entered properly? Also verify that you've got an adequate power supply, as brownouts can cause WiFi issues, at least 1A is recommended.

uli6, 2020-08-15:
> That's odd, I haven't come across that before. Are you certain your WiFi credentials are entered properly? Also verify that you've got an adequate power supply, as brownouts can cause WiFi issues, at least 1A is recommended.

Hi @jptrsn , 

Yes, the credentials are good and I've tried with different power supplies. :(

jptrsn, 2020-08-17:
Do you have any other devices you can test? It sounds like it may be an issue with the configuration file. Can you post the config entries (obviously, you should redact the secret info such as SSID, IP addresses, user names and passwords).

uli6, 2020-08-18:
Hi @jptrsn,

Found the problem! There was a different character in the SSID.

Thanks for your help.

jptrsn, 2020-08-18:
Glad to hear it's working now. 👍

closed 2020-08-18 by jptrsn