jptrsn/ESP32-mqtt-room #38: Add a white list of devices to reduce useless publishing on the MQTT server

PatBoud, 2020-12-19:
Hi! Thank you for this great tool, allows me to replace a RaspberryPi by an ESP-32 dev board! :)
I discovered that a lot of unwanted BLE devices are detected, like TV's, phones etc. I use 2 trackers, and 11 devices are detected by the scans.

A feature that may be very useful: 

Add a white list of BLE devices using their MAC address, in Settings.h .
Other BLE MAC addresses would simply be ignored, greatly reducing the number of publishes on the MQTT server.

Thank you!

PatBoud

PatBoud, 2020-12-19:
Well I decided to code it. So here's what I did.

In Settings.h, at the end:

```
//PatBoud - List of accepted MAC addresses for MQTT Publish. All others will be ignored.
#define whiteListCheck true
String whiteList[] = {"xxxxxxxxxxxx", "xxxxxxxxxxxx"};
uint32_t whiteListNumberOfItems = 2;
// --------------
```

And in ESP32-mqtt-room.ino  , in reportDevice function:

```
	...
	...
	mac_address.toLowerCase();

	//PatBoud - Check scanned MAC Address against WhiteList

	if (whiteListCheck) {
		bool whiteListFound = false;
		for (uint32_t x = 0; x < whiteListNumberOfItems; x++) {
			if (mac_address == whiteList[x]) {
				whiteListFound = true;
			}
		}

		if (whiteListFound == false) {
			return false;
		}
	}
	// --------------

```

It works fine for me! Hope it inspires you to implement it! :)

PatBoud

jptrsn, 2020-12-21:
Thanks for writing that - looks like a useful addition to the codebase. I'm actually no longer using this project myself, but if you want to submit a pull request with this integrated, I would be happy to review/merge.

maxi1134, 2021-01-17:
Hi Patboud,

Could you let me know at which lines these modifications go in ESP32-mqtt-room.ino?

Complete coving noob here!

PatBoud, 2021-01-23:
@jptrsn Hi! I sent a Pull Request with the new feature. Renamed it "Allowed list". Hope everything is all right!

@maxi1134 After `mac_address.toLowerCase();` within `bool reportDevice(BLEAdvertisedDevice advertisedDevice)` function, around line #320. Feature should be included soon, waiting for approval from the developer :)

jptrsn, 2021-01-25:
@PatBoud approved and merged.

closed 2021-01-25 by jptrsn