jptrsn/ESP32-mqtt-room #43: No devices are found

thehijacker, 2021-03-04:
**Describe the bug**
When doing scan, no devices are found.

**Configuration**
```
//Replace with your Wifi SSID; example: #define ssid "MyWifi"
#define ssid "WiFi"

//Replace with your Wifi password; example: #define password "12345678"
#define password "xxx"

//Replace with a human-friendly host name. Must not contain spaces or special characters and be unique on your network
#define hostname "esp32_room_presence"

//Replace with your MQTT Broker address; example: #define mqttHost IPAddress(192, 168, 1, 195)
#define mqttHost IPAddress(192, 168, x, xx)

//Replace with your MQTT Broker port; example: #define mqttPort 1883
#define mqttPort 1883

//Replace with your MQTT Broker user; example: #define mqttUser "homeassistant"
#define mqttUser "xx"

//Replace with your MQTT Broker password; example: #define mqttPassword "12345678"
#define mqttPassword "xx"

//Replace with the room name where the node will be placed; example: #define room "living-room"
#define room "kabinet"

//Specify the LED pin. For most dev boards, this is GPIO2
#define LED_BUILTIN 2

// Logic level for turning the led on. Most boards use active low, meaning LED_ON should be set to 0
#define LED_ON 0

//Define the base topic for room detection. Usually "room_presence"
#define channel "room_presence"

//Define the topic for publishing availability
#define availabilityTopic "presence_nodes/" room

//Define the topic for publishing JSON attributes
#define telemetryTopic "presence_nodes/" hostname "/tele"

// Define bluetooth scan parameters
#define scanInterval 5 // Define the interval in seconds between scans
#define singleScanTime 5 // Define the duration of a single scan in seconds
#define activeScan true // Active scan uses more power, but get results faster
#define bleScanInterval 0x80 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing
#define bleScanWindow 0x10 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing

// Maximum distance (in meters) to report. Devices that are calculated to be further than this distance in meters will not be reported
#define maxDistance 2

// MQTT topic for sensor values from HTU21D temperature and humidity sensor
//#define htuSensorTopic "presence_nodes/" hostname "/sensor"

//List of allowed MAC Addresses for MQTT Publish. All others will be ignored.
//Feature is disabled by default.
#define allowedListCheck false
String allowedList[] = {"11223344aabb", "11223344aabb"};
uint32_t allowedListNumberOfItems = 2;
```

**Logs**
```
Scanning...     Scan done! Devices found: 0
devices_discovered: 0
devices_reported: 0
Telemetry sent
```

**Hardware Details**
I have ESP32S developer board. Two actually. On one I am running OpenMQTTGateway that detects all my Xiaomi temperature sensors. For this second one I plan to put it in one room in which I am working from home so I can have specific automation working when I am inside. I have put OpenMQTTGateway on the same ESP32 and it found devices so it is working. But with ESP32-mqtt-room it finds none.

I can see device in MQTT and state connected. So WiFi and MQTT connection is working fine. I can also see blue LED on ESP32 blinking when it says Scanning...

**To Reproduce**

Connect to USB power and check serial logs.

**Expected behavior**

I expect to find many devices inside the house. I have Beacon Simulator on phone and expect for it to detect it also.

**Screenshots**

**Additional context**

thehijacker, 2021-03-04:
Tried to tweak the settings. Tried everything I could think of. Still get:

```
Scanning...     Scan done! Devices found: 0
devices_discovered: 0
devices_reported: 0
Telemetry sent
```

Anything else I can try to debug this?

RRRoamerr, 2021-03-04:
I've just had the same experience, I'm using ESP32-WROOM-32 boards, and it's not finding anything, physical beacon tags or simulated. Always 0 devices.

thehijacker, 2021-03-04:
OK. Fixed it. Replaced ESP32 BLE Arduino with NimBLE-Arduino (https://github.com/h2zero/NimBLE-Arduino). Looked at OpenMQTTGateway to see what library is used there and notice it is NimBLE also.

Maybe author of ESP32-mqtt-room can also swap it. There are minimal changes needed in .ino source.

```
Scanning...     Scan done! Devices found: 12
78597a85xxxx exceeded distance threshold 5.51
65e5e556xxxx exceeded distance threshold 5.58
5a8dc607xxxx exceeded distance threshold 4.72
a4c138d0xxxx exceeded distance threshold 3.65
a4c1383dxxxx exceeded distance threshold 6.07
devices_discovered: 12
devices_reported: 7
Telemetry sent
```

VanGog86, 2021-03-07:
> OK. Fixed it. Replaced ESP32 BLE Arduino with NimBLE-Arduino (https://github.com/h2zero/NimBLE-Arduino). Looked at OpenMQTTGateway to see what library is used there and notice it is NimBLE also. [...]

I have the same problem. please tell us more about how you solved it

thehijacker, 2021-03-08:
I was using Atom with PlatformIO addon. Under Libraries I have installed NimBLE-Arduino library.

I had to edit ESP32-mqtt-room.ino to use new library.

Line 25-36:

```
#include <AsyncTCP.h>
#include <NimBLEDevice.h>
//#include <NimBLEUtils.h>
//#include <NimBLEScan.h>
#include <NimBLEAdvertisedDevice.h>
#include <AsyncMqttClient.h>
#include <ArduinoJson.h>
#include <ArduinoOTA.h>
#include "NimBLEBeacon.h"
#include "NimBLEEddystoneTLM.h"
#include "NimBLEEddystoneURL.h"
```

Line 499: 

`void onResult(BLEAdvertisedDevice *advertisedDevice) {`

And it compiles just fine. But currently I am not using it. I found Raspberry Pi 4 at home and testing the room-assistant. So far it works better. It gives much faster result to Home Assistant how for the device is. With ESP32-mqtt-room this was not "instant".

Not sure if there is an option to make it work instantly.

maxi1134, 2021-03-10:
Same problem here. Will try your solution shortly. This was driving me mad, tried 3 ESP32 and 4 trackers.

jptrsn, 2021-03-11:
@thehijacker Thank you for discovering the problem. I actually no longer use this project, and just maintain it for others now. Can you submit a pull request with your changes that fix the problem? I can then merge those into the main project so others can benefit from your solution.

I will hopefully have some time this weekend to take a look.

artursmolarek, 2021-03-18:
I can recommend using this fork https://github.com/DTTerastar/ESP32-mqtt-room
The author resolved the issue. Besides that, it contains some new interesting features and improvements.

closed 2021-03-20 by jptrsn