jptrsn/ESP32-mqtt-room #45: DTTerastar fork: manual wifi ssid

OptimusGREEN, 2021-04-03:
@DTTerastar it would be great to be able to mannualy type ssid in web config as this would allow the use of hidden wifi.

maxi1134, 2021-04-03:
@OptimusGREEN  Did you managed to connect his fork to any of your wifis?

My ESP32 always opens up an AP on 192.168.4.1 even tho I give him the manual credentials. ( When using DTTerastar fork )

OptimusGREEN, 2021-04-03:
@maxi1134 Yes the manual config appears to be ignored but i can connect via the web config by selecting an ssid from the dropdown. I guess in the meantime it'll just be a case of making the wifi visible to connect then hide it again after connecting.

DTTerastar, 2021-04-04:
I'm not sure the https://github.com/Juerd/ESP-WiFiSettings component I picked will allow that.  You could report that to them and then we'll update to that once supported.

OptimusGREEN, 2021-04-04:
@DTTerastar ahh right nice one. I have done that now. https://github.com/Juerd/ESP-WiFiSettings/issues/16#issue-849873364

OptimusGREEN, 2021-04-05:
As you can see by the reply in the wifi settings repo, this isn't going to happen. Maybe just me but he came across a little rude

closed 2021-04-05 by OptimusGREEN

DTTerastar, 2021-04-05:
> As you can see by the reply in the wifi settings repo, this isn't going to happen. Maybe just me but he came across a little rude

Agreed, he was a little harsh.  We could pick another one of the many Wifi auto configure portal libs if it's really needed.