jptrsn/ESP32-mqtt-room #46: Not seeing any BLE devices

dbond82, 2021-04-08:
Hi,

I have successfully uploaded your project into one of my ESP32 boards for use with my Home assistant but it is not seeing any BLE devices.  I have many iBeacon units as well as units that would send just a mac address:

I have tried hardcoding a MAC address into the settings file and setting the filter to true but still nothing
settings.h file extract all other settings set for my network and mqtt access:

#define allowedListCheck true
String allowedList[] = {"AC233F58A884", "11223344aabb"};
uint32_t allowedListNumberOfItems = 2;

The mac is not important so I have not "hidden" it from your view

Output from serial monitor:
Scanning...     Scan done! Devices found: 0
devices_discovered: 0
devices_reported: 0
Telemetry sent

All the MQTT links and WiFi connection etc is correct but I just can't seem to work out why I can't see BLE devices I probably have 10-20 in the house with battery on as I use them at work for other projects just wanted to use tracking in the house too.

I have tried using Estimote Beacon Minew and using my Phone (Android) with an iBeacon/Eddystone client app running

I have not used VC studio before with platformio as I always use Arduino IDE or the home Assistant ESPHome for very basic projects normally (temp lights etc) so I may be doing something wrong but everything compiles and uploads so I would have thought I have the required libraries etc.  I am hoping you could point me to some setting I have overlooked.

Apologies if I am missing something obvious and many thanks in advance

Regards

Dave

jptrsn, 2021-04-08:
Can you try removing the allowed MAC addresses and have it report everything it encounters? It may be an issue with the format of the MAC addresses being broadcast in the advertisement. Once you do that, you'll be able to see the devices that get reported by subscribing to the MQTT topic, and viewing the messages that get sent. That should allow you to find the right MAC address.

The only reason to use the allowed list is to reduce the number of messages sent to the MQTT server. Allowing it to send all discovered devices shouldn't impact performance of your MQTT server much (unless it's running slowly to begin with), and would enable you to include new devices in your home assistant config without uploading a new build of the firmware.

I no longer use this project, so it is not maintained or monitored for functionality. The allowed list is something I have not used myself, but was contributed by another user.

dbond82, 2021-04-08:
Hi Thanks for the speedy response.  Apologies I was not clear that I did originally not have the filter turned on but tried it in case it made a difference.  I have removed all this now but still the same.  Its weird as I know the ESP board can see BLE when I upload a different code from Arduino IDE BLE_scan_mqtt but this does not have the feature set of your code with the Home assistant.  Sorry I have tried loads of bits since you replied I wonder if my VScode is not correct as it was a new install and I had lots of red squiggles regarding the various libraries but it does compile and upload so I would have thought this is all resolved??  Any ideas?  sorry to bother you further.

Dave

dbond82, 2021-04-08:
I fixed it I had to add a few more BLE libraries thanks for the support.

Dave

closed 2021-04-08 by dbond82

micahriley88, 2021-04-09:
> I fixed it I had to add a few more BLE libraries thanks for the support.
> 
> Dave

Could you perhaps clarify on which BLE libraries you added to solve this problem? 

I'm having the same issue with 0 devices showing for discovery/reported.

dbond82, 2021-04-09:
Sure no problem.

I added:
ESP32 BLE Arduino
ArduinoJSON
AsyncMqttCLient
AsyncTCP
AsyncTCP-esphome
ESPAsyncTCP
NTPClient

Not all of these libraries were needed for this project but I was downloading some for another (NTP etc) I believe it was the BLE Arduino that resolved the issue.

I downloaded them directly in platformio section of VSCode and associated them to the project so within my project folder (on Windows Explorer) I have a .pio/libdeps/esp32 and the list above as folders within.

Hope this helps