jptrsn/ESP32-mqtt-room #47: Updated Settings for a ESP32 and it no longer detects info for Home Assistant

seangreen2, 2021-06-13:
**Describe the bug**
For about the past year I have been using ESP32-mqtt-room for room presence and it has worked flawlessly when I set it up the first time. I wanted to make a few minor changes in range detection on one of the ESP32s, but once I flashed the new data on the chip, Home Assistant no longer "sees" data coming from it despite it being connected to my MQTT Broker.

I have troubleshooted everything I can think of, but nothing I do seems to work and I'm hoping someone here knows way more about MQTT that may be able to offer some insight.

I have been using PlatformIO on VScode to flash the ESP32s without issue. I'm also using a Beacon Simulator on my android phone to be tracked by the ESP32s which has worked without issue as well.

- First, here's my proof that the ESP32 in question is connecting to the broker: https://i.imgur.com/7pme7Cf.png
- Here's my current settings for my MQTT integration: https://i.imgur.com/YN1KUTP.png
- Here's a link to the github for ESP32-mqtt-room: https://github.com/jptrsn/ESP32-mqtt-room
- Here's the Settings_locahl.h parameters I am using to flash to the ESP32: https://pastebin.com/6pkQWm6T
- And lastly, here's how the sensor appears in Home Assistant:

```
  - platform: mqtt_room
    device_id: "9ad13236f8064c4caa1d1be47d75e12e-1-0"
    name: 'Phone Room Location'
    state_topic: 'room_presence'
    timeout: 10
    away_timeout: 15
```

I've tried going back a few versions on github, I have also tried using older versions of PlatformIO, have uninstalled/reinstalled the MQTT integration, created a new beacon on my phone, but nothing. I am at a complete loss and this is as far as my knowledge goes.

If there's any other information I'm able to provide, please let me know. I would hate to lose out on this great solution to Room Presence.

**Configuration**
```
//Replace with your Wifi SSID; example: #define ssid "MyWifi"
#define ssid "Secret SSID"
 
//Replace with your Wifi password; example: #define password "12345678"
#define password "mypassword"
 
//Replace with a human-friendly host name. Must not contain spaces or special characters and be unique on your network
#define hostname "living_room_presence"
 
//Replace with your MQTT Broker address; example: #define mqttHost IPAddress(192, 168, 1, 195)
#define mqttHost IPAddress(192, 168, 1, 185)
 
//Replace with your MQTT Broker port; example: #define mqttPort 1883
#define mqttPort 1883
 
//Replace with your MQTT Broker user; example: #define mqttUser "homeassistant"
#define mqttUser "homeassistant"
 
//Replace with your MQTT Broker password; example: #define mqttPassword "12345678"
#define mqttPassword "xxxxxxxxxxxxxxxxxxxxxxxx"
 
//Replace with the room name where the node will be placed; example: #define room "living-room"
#define room "living_room"
 
//Specify the LED pin. For most dev boards, this is GPIO2
#define LED_BUILTIN 2
 
// Logic level for turning the led on. Most boards use active low, meaning LED_ON should be set to 0
#define LED_ON 0
 
//Define the base topic for room detection. Usually "room_presence"
#define channel "room_presence"
 
//Define the topic for publishing availability
#define availabilityTopic "presence_nodes/" room
 
//Define the topic for publishing JSON attributes
#define telemetryTopic "presence_nodes/" hostname "/tele"
 
// Define bluetooth scan parameters
#define scanInterval 5 // Define the interval in seconds between scans
#define singleScanTime 5 // Define the duration of a single scan in seconds
#define activeScan true // Active scan uses more power, but get results faster
#define bleScanInterval 0x80 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing
#define bleScanWindow 0x10 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing
 
// Maximum distance (in meters) to report. Devices that are calculated to be further than this distance in meters will not be reported
#define maxDistance 7
 
// MQTT topic for sensor values from HTU21D temperature and humidity sensor
//#define htuSensorTopic "presence_nodes/" hostname "/sensor"
 
 
//List of allowed MAC Addresses for MQTT Publish. All others will be ignored.
//Feature is disabled by default.
#define allowedListCheck false
String allowedList[] = {"11223344aabb", "11223344aabb"};
uint32_t allowedListNumberOfItems = 2;
```

**Logs**
I don't have any logs from the serial, but I used MQTT Explorer to find that the ESP32 is sending this to the broker:
`{"room":"living_room","ip":"192.168.1.163","hostname":"living_room_presence","scan_dur":5,"wait_dur":5,"max_dist":7,"disc_ct":0,"rept_ct":0}`

Here's a image of how I got this information: https://i.imgur.com/Y2dSALz.png

**Hardware Details**
Using Beacon Simulator for Android. Has worked fine up until I attempted to make changes to my ESP32 settings.

jptrsn, 2021-06-15:
The message you're seeing in MQTT is the telemetry. This essentially reports the status of the device, rather than any scan results, so good news: the device is connected and working.

You'll want to watch the room assistant topic itself to check for any device scan results. Take a look at the docs for troubleshooting advice.

https://jptrsn.github.io/ESP32-mqtt-room/troubleshooting

The section you're looking for is the last one, Home Assistant (purple arrow).

seangreen2, 2021-06-16:
Appreciate the reply. Good to know that everything is connecting just fine.

I did what you said and found something to subscribe to the `room_presence/#` topic. Unfortunately it appears that nothing is simply being published to that particular topic and I kept the default settings.

https://i.imgur.com/njqCQCd.png

I also subscribed to # to ensure I was at least getting some information and it began to display the info I already had before of the telemetry.

https://i.imgur.com/OTnQ2RK.png

Any ideas as to why nothing is getting published to the `room_presence` topic?

Subcode, 2021-06-16:
```
"disc_ct":0,
"rept_ct":0
```

This means no discovered devices and no reported devices right. So its not detecting anything with the scan.
I'm having the same issue which led me here. I know for a fact there are BLE devices but the ESP32s doesnt seem to see anything to report.

seangreen2, 2021-06-22:
Just bumping in hopes that a solution may be found. Afraid I hit a wall again and am unable to proceed. This is some incredible software for room presence and I would hate to lose it.

Subcode, 2021-06-26:
It might be related to this problem.
https://stackoverflow.com/questions/66755676/ble-scanner-device-found-0

The fix seems to be using version 1.0.4 of the ble scanning library.

`Devices count always 0: This is due to a change in operation of tracking the scan results. An obvious change to the operation is that when a BLE scan operation has user scanning callbacks in place, the tracking of scanned devices is disabled (which is why the count is always 0)! Even though this is obvious in the source code (a comment states this), it is not obvious to the library user as the examples using BLE scanning do not mention this.`

Subcode, 2021-06-26:
Fixed it.
Disable this line in ESP32-mqtt-room.ini:
`pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());`

Because of the library update this disables the ability to count devices which this software depends on.
The callback is used to turn the led on and off so its completely useless functionally.

seangreen2, 2021-06-26:
You legend! That fixed the issue and everything is getting connected just fine again!

Thanks for helping out and getting this back up and running.

jptrsn, 2021-06-26:
Thanks @seangreen2 and @Subcode for fixing this problem. I've tested it locally, and it appears to be working for me, so I've created a new release with that minor change, and a slight improvement to `platformio.ini` to better make use of VSCode.

closed 2021-06-26 by jptrsn