jptrsn/ESP32-mqtt-room #48: Link to Person in HomeAssistant

Subcode, 2021-07-03:
I am trying to link the mqtt-room tracker to a person in home assistant so i can track if a person is home / in what room within home assistant. But there doesnt seem to be a device_tracker link in mqtt-room to enable this with.
It'd be great if i could link a beacon to a person. Preferably through a device tracker of some kind so i can add it through the home assistant person interface.
It would allow for a lot more automation.

anarchking, 2021-07-03:
Yeah its not easy but I use templates to accomplish that.

Subcode, 2021-07-03:
Could you post the template here?

anarchking, 2021-07-03:
- platform: mqtt_room
  device_id: xx:xx:xx:xx:xx:xx
  name: AnarchKing_is_in
  state_topic: "home/home_presence"
  #timeout:
  #away_timeout:

I name the sensors the name of the room and give it a number cuz I sometimes have multiple sensors in one room.
So sensor will display (AnarchKing_is_in OfficeBLE1)