jptrsn/ESP32-mqtt-room #50: Can't get OTA to work

soloam, 2021-07-22:
**Describe the bug**
I tried several ways to OTA my esp32, but never works, I use VSCode, but also tried to upload by command line, I allways get a "No response from the ESP" error

**Configuration**
```
//Replace with your Wifi SSID; example: #define ssid "MyWifi"
#define ssid "WiFi"

//Replace with your Wifi password; example: #define password "12345678"
#define password "xxxxxx"

//Replace with a human-friendly host name. Must not contain spaces or special characters and be unique on your network
#define hostname "esp32_rp_bedroom1"

//Replace with your MQTT Broker address; example: #define mqttHost IPAddress(192, 168, 1, 195)
#define mqttHost "mosquitto.home"

//Replace with your MQTT Broker port; example: #define mqttPort 1883
#define mqttPort 1883

//Replace with your MQTT Broker user; example: #define mqttUser "homeassistant"
#define mqttUser ""

//Replace with your MQTT Broker password; example: #define mqttPassword "12345678"
#define mqttPassword ""

//Replace with the room name where the node will be placed; example: #define room "living-room"
#define room "bedroom"

//Specify the LED pin. For most dev boards, this is GPIO2
#define LED_BUILTIN 2

// Logic level for turning the led on. Most boards use active low, meaning LED_ON should be set to 0
#define LED_ON 0

//Define the base topic for room detection. Usually "room_presence"
#define channel "room_presence"

//Define the topic for publishing availability
#define availabilityTopic "presence_nodes/" hostname "/connected"

//Define the topic for publishing JSON attributes
#define telemetryTopic "presence_nodes/" hostname "/tele"

// Define bluetooth scan parameters
#define scanInterval 5 // Define the interval in seconds between scans
#define singleScanTime 5 // Define the duration of a single scan in seconds
#define activeScan true // Active scan uses more power, but get results faster
#define bleScanInterval 0x80 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing
#define bleScanWindow 0x10 // Used to determine antenna sharing between Bluetooth and WiFi. Do not modify unless you are confident you know what you're doing

// Maximum distance (in meters) to report. Devices that are calculated to be further than this distance in meters will not be reported
#define maxDistance 0

// MQTT topic for sensor values from HTU21D temperature and humidity sensor
//#define htuSensorTopic "presence_nodes/" hostname "/sensor"

//List of allowed MAC Addresses for MQTT Publish. All others will be ignored.
//Feature is disabled by default.
#define allowedListCheck false
String allowedList[] = {"123", "456"};
uint32_t allowedListNumberOfItems = 2;

```

**Logs**
```
CURRENT: upload_protocol = esptool
Uploading .pio/build/esp32/firmware.bin
11:32:19 [DEBUG]: Options: {'esp_ip': '192.168.7.107', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 43215, 'auth': '', 'image': '.pio/build/esp32/firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
11:32:19 [INFO]: Starting on 0.0.0.0:43215
11:32:19 [INFO]: Upload size: 1433984
Sending invitation to 192.168.7.107 ..........
11:33:59 [ERROR]: No response from the ESP
*** [upload] Error 1
============================================== [FAILED] Took 145.08 seconds ==============================================
The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1.
```

I can ping the device and the device is reporting measurements into MQTT

jptrsn, 2021-07-22:
When was the last time you updated this device successfully OTA? At one point, PlatformIO changed the default OTA port on the device from `32` to `3232`, which could potentially be the reason you're seeing the error, or it could be entirely unrelated.

Try adding the following to your environment configuration in platformio.ini (from [docs linked here](https://docs.platformio.org/en/latest/platforms/espressif32.html#authentication-and-upload-options))
```
upload_flags =
    --port=32
```

If that doesn't work, then I'm not sure how to help, as I no longer use this project and do not have any devices running the code.

soloam, 2021-07-22:
Same problem! I never was able to OTA update any of my esp32! So you abandoned this project? No more updates expected?

Tks for your work

jptrsn, 2021-07-23:
It feels like it could be a network issue, I'm really not sure how to help beyond this point. You're correct, I'm not planning on updating this repository any more. I think if anything, it would make more sense to develop a component that would integrate into ESP Home.

closed 2021-07-23 by jptrsn