jptrsn/ESP32-mqtt-room #55: Team Up - DHT 11/22

jonathanicq, 2022-01-22:
** First of all, thanks for your project... I've only read since I'm on a business trip, but it will be the first thing when I get back home **

Would be nice if we could keep using the EPS for other sensors,
since the idea (at least my idea) is to have one for each room at home and I want to do the same for a temperature and humidity sensor would be easier if both could run on the same board.

maxi1134, 2022-01-22:
You obviously didn't read, since the first thing it says in the README, is that this project is pretty much dead.

jonathanicq, 2022-01-22:
Thanks @maxi1134 now I feel stupid :)
I fact I messed up with the youtube video from The Smart Home where he mention that he is using this project.
So I read everythinf there and jump to conclusions... Anyways, thanks for you comment

jptrsn, 2022-01-24:
@jonathanicq check out the espresence project, which forked this code base and has evolved quite a bit since then. It really looks like a fantastic solution. https://espresense.com/

closed 2022-01-24 by jptrsn